monome/norns-image #49: GPIO Pins incorrect?

okyeron, 2018-06-11:
GPIO Pins are listed here:
https://github.com/monome/norns-image/blob/master/readme-hardware.md

This is confusing as I think what's listed is the actual pin # rather than the GPIO#

Also - Pin 39 (GPIO39 as listed) is GND

Are these the correct assignments? Where is the dt_overlay for these?

Here's a reference image showing pin number as well as GPIO#
[image]

okyeron, 2018-06-11:
Aha! That certainly explains my confusion.

Hoping there is a way to map the GPIOs to a RasPI 3 sometime soon. Also - I'm happy to help with testing or whatever might needed there.

closed 2018-06-11 by okyeron

tehn, 2018-06-11:
> does seem like there are some errors (e.g., GPIO02 and GPIO03 alt funcs are i2c0, not i2c1)

just checked the doc, it lists i2c0? didn't see any errors, checked against schem just now but please let me know if you see anything wrong

catfact, 2018-06-11:
sorry sorry sorry typo, i meant the other way round. 
doc says
```The i2c lines are connected to i2c0:
    PIN: SDA = GPIO2
    PIN: SCL = GPIO3
```

datasheet table 9.1.1 alt funcs says
```
GPIO0 - SDA0
GPIO1 - SCL0
GPIO2 - SDA1
GPIO3 - SCL1
```