monome/norns-image #52: build instructions for raspberry pi 3

tehn, 2018-07-04:
need build guide for off-the-shelf pi 3

okyeron, 2018-07-05:
How best to submit this?

Would we want to include compile the kernel from source or using an image?

tehn, 2018-07-05:
just a README would suffice, capturing the tips posted in the lines thread.

generally the existing norns-specific README is pretty complete, so this would just need to be an addendum for rpi3 specific issues.

simonvanderveldt, 2018-07-05:
I believe I've said the same in the topic on lines, but just use the upstream/Raspberry kernel. If you want to compile the kernel yourself use the upstream sources.

okyeron, 2018-07-05:
Yeah - I'm an outlier since I was trying to compile in support for the ssd1322 display - but that had some issues compiling that on the most recent stretch releases. Raspi people looking for a 2.7in display should probably use a ssd1305 since that's already in the kernel.

ngwese, 2018-07-05:
@okyeron - this might be of interest: https://github.com/monome/linux/commit/b63261e99e4716c61b4d1bb3d09470ad53a5d23e

okyeron, 2018-07-05:
Yes indeed. Thanks. I'd ended up rewriting that myself (not sure if I was doing it correctly) :)

FigrHed, 2018-07-10:
From the thread it sounded like one would have to make some hardware decisions as there aren't enough GPIO pins for the screen, the sound card and the controls. Is this correct?

okyeron, 2018-07-10:
With a stock Raspberry Pi there are plenty of GPIO pins available to account for a soundcard, display and 3 encoders/buttons. You may be making reference to a comment that the PiSound soundcard/DAC may not have enough available pins (?). 

Pin availability will depend on what soundcard you use, so you'd need to do some research.

jlmitch5, 2018-07-10:
Could we either expand the scope of this issue or open a new issue for documentation as it relates to building norns-image on a linux desktop, as well as for building for virtualization (i.e. docker or vagrant for mac/windows users)?  Those are the ones I've seen mentioned the most on the forum.

I also remember reading some things about real time weirdness under virtualization.  Could be good to include any limitations running on these other types of platforms has.

okyeron, 2018-07-12:
Should the raspi instructions include USB Audio setup instructions?

sinitsinmike, 2024-10-02:
Is there a guide anywhere on how to use a different display/screen. It is hard to fing the specified part.
Is there a way to use some other display via SPI or I2C?

tehn, 2024-10-02:
try octopart. they're widely available.

closed 2024-10-02 by tehn

sinitsinmike, 2024-10-02:
Thanks! Just wondering if anyone tried to build Norns Monome code on Git running a different screen? Sinse it's open source. I bet someone tried it.
That screen isn't really available on the islands where I live. But many other screens from Aliexpress are available. That's why I'm asking ;)