monome/norns-image #66: CM3+ temperature measurement not working

simonvanderveldt, 2019-03-14:
When running the 190303 beta image on a CM3+ I get the following error message when reading the CPU temperature
```
we@norns:~$ vcgencmd measure_temp
VCHI initialization failed
we@norns:~$ echo $?
255
```
This is probably caused by the firmware not matching with the `vcgencmd` binary.
Need to figure out how to update both the firmware and the other stuff from https://github.com/raspberrypi/firmware without updating the kernel.

ngwese, 2019-03-18:
I can't reproduce this. I upgraded to a CM3+ using the 190314 image and temperature reading seems to work (both as root and not):
```
we@norns (~) 500% sudo -s
root@norns:/home/we# vcgencmd measure_temp
temp=41.9'C
root@norns:/home/we# exit
we@norns (~) 501% vcgencmd measure_temp
temp=42.4'C
```

simonvanderveldt, 2019-03-21:
Yeah, it's working for me as well on the 190314 image. I think it's because I copied some other files than @tehn has copied from the raspberry firmware repo (had to copy the myself into the 190303 image because otherwise the CM3+ won't boot).

We'll have to determine how we want to update the firmware related stuff (that's all the firmware related files in `/boot`, but also the tools like `vcgencmd` and the libs to make these tools work. There are packages in raspbian for them:
```
dpkg -L raspberrypi-bootloader
/.
/boot
/boot/LICENCE.broadcom
/boot/bootcode.bin
/boot/fixup.dat
/boot/fixup_cd.dat
/boot/fixup_db.dat
/boot/fixup_x.dat
/boot/start.elf
/boot/start_cd.elf
/boot/start_db.elf
/boot/start_x.elf
/usr
/usr/share
/usr/share/doc
/usr/share/doc/raspberrypi-bootloader
/usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz
/usr/share/doc/raspberrypi-bootloader/copyright
we@norns:~$ dpkg -L libraspberrypi-bin
/.
/opt
/opt/vc
/opt/vc/bin
/opt/vc/bin/containers_check_frame_int
/opt/vc/bin/containers_datagram_receiver
/opt/vc/bin/containers_datagram_sender
/opt/vc/bin/containers_dump_pktfile
/opt/vc/bin/containers_rtp_decoder
/opt/vc/bin/containers_stream_client
/opt/vc/bin/containers_stream_server
/opt/vc/bin/containers_test
/opt/vc/bin/containers_test_bits
/opt/vc/bin/containers_test_uri
/opt/vc/bin/containers_uri_pipe
/opt/vc/bin/dtmerge
/opt/vc/bin/dtoverlay
/opt/vc/bin/dtoverlay-post
/opt/vc/bin/dtoverlay-pre
/opt/vc/bin/edidparser
/opt/vc/bin/mmal_vc_diag
/opt/vc/bin/raspistill
/opt/vc/bin/raspivid
/opt/vc/bin/raspividyuv
/opt/vc/bin/raspiyuv
/opt/vc/bin/tvservice
/opt/vc/bin/vcdbg
/opt/vc/bin/vcgencmd
/opt/vc/bin/vchiq_test
/opt/vc/bin/vcmailbox
/opt/vc/bin/vcsmem
/opt/vc/sbin
/opt/vc/sbin/vcfiled
/usr
/usr/bin
/usr/share
/usr/share/doc
/usr/share/doc/libraspberrypi-bin
/usr/share/doc/libraspberrypi-bin/LICENCE
/usr/share/doc/libraspberrypi-bin/changelog.Debian.gz
/usr/share/doc/libraspberrypi-bin/copyright
/opt/vc/bin/dtparam
/usr/bin/dtmerge
/usr/bin/dtoverlay
/usr/bin/dtoverlay-post
/usr/bin/dtoverlay-pre
/usr/bin/dtparam
/usr/bin/edidparser
/usr/bin/raspistill
/usr/bin/raspivid
/usr/bin/raspividyuv
/usr/bin/raspiyuv
/usr/bin/tvservice
/usr/bin/vcdbg
/usr/bin/vcgencmd
/usr/bin/vchiq_test
```

So we can update them via `apt-get` but there's no way to to guarantee idempotency for the result since we can't fix the version of these packages this way meaning at a different point in time you'll get a different result, no a good thing for guaranteeing working results.
On the other hand I don't really want to be managing/installing these files manually or via scripts, so maybe our best bet is to mirror the current version of them (which seems to be `1.20190215-1` for all of them at the moment) and install them that way? And then periodically update them in tandem with the kernel.

simonvanderveldt, 2019-04-08:
This has been fixed in the 190314 image

simonvanderveldt, 2019-04-09:
@tehn this is unfortunately broken on the new 190405 image.
Do we want to go for a quick fix? (which would be to copy the contents from the boot partition from the 190314 image that one was working correctly)
Otherwise we can figure out the firmware thing in the other issue and it'll be solved that way.

tehn, 2019-04-09:
copying from earlier version is fine. confusing because I am pretty sure I copied the same files!

ideally these could get copied into a kernel release tgz?

PR #76 (2019-05-04): Install/update raspberry firmware files using a specific version
This installs a specific version of the firmware files. Apart from hosting these files ourselves and having multiple versions in the `Package` file in our repo this is the only way to guarantee a certain version of the firmware files and the accompanying tools is installed for a specific norns update.

I'm not super happy with this solution but I don't think there's an easier way to guarantee a certain version of the firmware files is installed.

Alternatively we could just always install the latest version, in theory newer versi [... cut]
closed 2019-05-04 by simonvanderveldt