monome/norns-image #74: Determine how to install/update raspberry firmware files

simonvanderveldt, 2019-04-08:
Followup from #66 

We'll have to determine how we want to update the firmware related stuff (that's all the firmware related files in `/boot`, but also the tools like `vcgencmd` and the libs to make these tools work. There are packages in raspbian for them:
```
dpkg -L raspberrypi-bootloader
/.
/boot
/boot/LICENCE.broadcom
/boot/bootcode.bin
/boot/fixup.dat
/boot/fixup_cd.dat
/boot/fixup_db.dat
/boot/fixup_x.dat
/boot/start.elf
/boot/start_cd.elf
/boot/start_db.elf
/boot/start_x.elf
/usr
/usr/share
/usr/share/doc
/usr/share/doc/raspberrypi-bootloader
/usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz
/usr/share/doc/raspberrypi-bootloader/copyright
we@norns:~$ dpkg -L libraspberrypi-bin
/.
/opt
/opt/vc
/opt/vc/bin
/opt/vc/bin/containers_check_frame_int
/opt/vc/bin/containers_datagram_receiver
/opt/vc/bin/containers_datagram_sender
/opt/vc/bin/containers_dump_pktfile
/opt/vc/bin/containers_rtp_decoder
/opt/vc/bin/containers_stream_client
/opt/vc/bin/containers_stream_server
/opt/vc/bin/containers_test
/opt/vc/bin/containers_test_bits
/opt/vc/bin/containers_test_uri
/opt/vc/bin/containers_uri_pipe
/opt/vc/bin/dtmerge
/opt/vc/bin/dtoverlay
/opt/vc/bin/dtoverlay-post
/opt/vc/bin/dtoverlay-pre
/opt/vc/bin/edidparser
/opt/vc/bin/mmal_vc_diag
/opt/vc/bin/raspistill
/opt/vc/bin/raspivid
/opt/vc/bin/raspividyuv
/opt/vc/bin/raspiyuv
/opt/vc/bin/tvservice
/opt/vc/bin/vcdbg
/opt/vc/bin/vcgencmd
/opt/vc/bin/vchiq_test
/opt/vc/bin/vcmailbox
/opt/vc/bin/vcsmem
/opt/vc/sbin
/opt/vc/sbin/vcfiled
/usr
/usr/bin
/usr/share
/usr/share/doc
/usr/share/doc/libraspberrypi-bin
/usr/share/doc/libraspberrypi-bin/LICENCE
/usr/share/doc/libraspberrypi-bin/changelog.Debian.gz
/usr/share/doc/libraspberrypi-bin/copyright
/opt/vc/bin/dtparam
/usr/bin/dtmerge
/usr/bin/dtoverlay
/usr/bin/dtoverlay-post
/usr/bin/dtoverlay-pre
/usr/bin/dtparam
/usr/bin/edidparser
/usr/bin/raspistill
/usr/bin/raspivid
/usr/bin/raspividyuv
/usr/bin/raspiyuv
/usr/bin/tvservice
/usr/bin/vcdbg
/usr/bin/vcgencmd
/usr/bin/vchiq_test
```

So we can update them via `apt-get` but there's no way to to guarantee idempotency for the result since we can't fix the version of these packages. In other words at a different point in time you'll get a different result, not a good thing for guaranteeing working results.

On the other hand I don't really want to be managing/installing these files manually or via scripts, so maybe our best bet is to mirror the `.deb` packages for the current version of them (`1.20190215-1` for all of them at the moment) and install them that way? And then periodically update them in tandem with the kernel.

artfwo, 2019-04-09:
But we can freeze particular versions of the packages, it's called holding: https://help.ubuntu.com/community/PinningHowto#Introduction_to_Holding_Packages

simonvanderveldt, 2019-04-09:
Yeah, but there's no way to guarantee that will give a specific version. Ie run the update that pins now vs in a months time.

simonvanderveldt, 2019-04-13:
So it seems like the Raspberry/raspbian repos keep old versions [around](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/), but only one version is listed in the [Package file](http://archive.raspberrypi.org/debian/dists/stretch/main/binary-armhf/Packages) (search for `Package: raspberrypi-bootloader` which means we can't install older versions using the normal command-line tools :|

PR #76 (2019-05-04): Install/update raspberry firmware files using a specific version
This installs a specific version of the firmware files. Apart from hosting these files ourselves and having multiple versions in the `Package` file in our repo this is the only way to guarantee a certain version of the firmware files and the accompanying tools is installed for a specific norns update.

I'm not super happy with this solution but I don't think there's an easier way to guarantee a certain version of the firmware files is installed.

Alternatively we could just always install the latest version, in theory newer versi [... cut]
closed 2019-05-04 by simonvanderveldt