monome/norns-image #81: broken display & controls on 200424

tridae, 2020-09-05:
I did flash the `200424` image. I could boot it successfully in the monome shield (200323)
However:
* the display was flipped (upside down)
* the encoder knobs have no effect
* the buttons have no effect
* "awake" does not play on startup

Is this a common bug or only on my shield? ("200106" did work fine)

tehn, 2020-09-05:
apologies, i mistakenly forgot that 200424 is only for norns, not shield. use the 200106-shield file as it has the corresponding kernel/etc

closed 2020-09-05 by tehn

tridae, 2020-09-05:
thank you for the clarification. You need to purge [my PR](https://github.com/monome/norns-shield/pull/3) then. I did it before testing the image. My mistake :-/