monome/norns-image #99: daemon.log shouldn't log OSC modulation

ambv, 2022-06-18:
Currently ws-wrapper logs every message received through OSC which might become a large amount of data if this is automated modulation. Case in point, after making a [MIDI to Norns script parameter automation M4L device](https://llllllll.co/t/midi-to-norns-script-parameter-automation-m4l-device/56274), I set it up automated via Ableton Live as part of longterm testing of [a slightly overclocked Norns](https://llllllll.co/t/adventures-in-overclocking-stock-norns/56239).

After around 24 hours of runtime, I can see `daemon.log` being 350MB big (!!!) with
* 2.5 million occurrences of `norns ws-wrapper[512]: setting parameter /release to ...`
* 1.25 million occurrences of `norns ws-wrapper[512]: setting parameter /pw to ...`
* 284 thousand occurrences of `norns ws-wrapper[512]: setting parameter /cutoff to ...`
* 181 thousand occurrences of `norns ws-wrapper[512]: setting parameter /delay_rate to ...`
* and so on, and so on

You get my point.

I understand that:
* people are unlikely to run Norns for 24+ hours at a time; and
* [logrotate is supposed to compress old logs](https://github.com/monome/norns-image/issues/96)

However, I think that:
* 24 sessions of 1 hour will end up with the same amount of log data; and
* writing this information to disk shortens the lifespan of the eMMC which is not replaceable without replacing the entire compute module which is [not easily obtainable these days](https://llllllll.co/t/scarcity-and-precarity-in-small-manufacturing/55896/44)

Therefore, I think ws-wrapper shouldn't log this activity unless instructed via sending it `/debug 1` or some Unix signal.

ambv, 2022-06-18:
By the way, the daemon.log information is also written to the journal of the `we` user. Something is wrong with that both because it's weird to have this data duplicated, but also because in `/etc/systemd/journald.conf` the config specifies that `SystemMaxUse=10M` which means the system shouldn't be writing more than 10M of logs to disk. However, this is what's actually happening:
```
root@norns:~# ls -lah /var/log/journal/8c4778817d5b4fad92166a8840486cc8/
total 316M
drwxr-sr-x+ 2 root systemd-journal 4.0K Jun 18 23:41 .
drwxr-sr-x+ 3 root systemd-journal 4.0K Jun 15 13:04 ..
-rw-r-----+ 1 root systemd-journal 8.0M Jun 18 01:17 system@0005e1adca309e18-9cd5d11aa357269c.journal~
-rw-r-----+ 1 root systemd-journal 8.0M Jun 18 23:41 system@0005e1c091f5a7db-ff1dac9073939963.journal~
-rw-r-----+ 1 root systemd-journal 8.0M Jun 17 20:20 system@8ae22133f9cc44ce99f6ab56b3cb109c-00000000003d1de5-0005e1a999591d5c.journal
-rw-r-----+ 1 root systemd-journal 8.0M Jun 17 20:55 system@8ae22133f9cc44ce99f6ab56b3cb109c-00000000003ed83f-0005e1aa197ecd14.journal
-rw-r-----+ 1 root systemd-journal 8.0M Jun 17 21:20 system@8ae22133f9cc44ce99f6ab56b3cb109c-00000000003f9fae-0005e1aa51d006dd.journal
-rw-r-----+ 1 root systemd-journal 8.0M Jun 17 22:07 system@8ae22133f9cc44ce99f6ab56b3cb109c-000000000041c7f8-0005e1aaf01272b3.journal
-rw-r-----+ 1 root systemd-journal 8.0M Jun 18 23:52 system.journal
-rw-r-----+ 1 root systemd-journal  40M Jun 18 01:17 user-1000@0005e1adca5906e3-f891a0c5f990e47e.journal~
-rw-r-----+ 1 root systemd-journal  43M Jun 17 20:37 user-1000@4b2cba08b7e3427cb654f7705aad78ab-00000000003cf074-0005e1a98d82ec5e.journal
-rw-r-----+ 1 root systemd-journal  43M Jun 17 20:59 user-1000@4b2cba08b7e3427cb654f7705aad78ab-00000000003e0bb8-0005e1a9e0d0c8d5.journal
-rw-r-----+ 1 root systemd-journal  43M Jun 17 21:22 user-1000@4b2cba08b7e3427cb654f7705aad78ab-00000000003f27d5-0005e1aa2ffc3261.journal
-rw-r-----+ 1 root systemd-journal  43M Jun 17 21:44 user-1000@4b2cba08b7e3427cb654f7705aad78ab-0000000000404335-0005e1aa8351b26f.journal
-rw-r-----+ 1 root systemd-journal  43M Jun 17 22:07 user-1000@4b2cba08b7e3427cb654f7705aad78ab-0000000000415f3e-0005e1aad27cde4d.journal
-rw-r-----+ 1 root systemd-journal 8.0M Jun 18 23:52 user-1000.journal
```

Looking closer at the system journals, we see this:
```
root@norns:~# file /var/log/journal/8c4778817d5b4fad92166a8840486cc8/system*
/var/log/journal/8c4778817d5b4fad92166a8840486cc8/system@0005e1adca309e18-9cd5d11aa357269c.journal~:                                 Journal file, offline
/var/log/journal/8c4778817d5b4fad92166a8840486cc8/system@0005e1c091f5a7db-ff1dac9073939963.journal~:                                 Journal file, offline
/var/log/journal/8c4778817d5b4fad92166a8840486cc8/system@8ae22133f9cc44ce99f6ab56b3cb109c-00000000003d1de5-0005e1a999591d5c.journal: Journal file, offline
/var/log/journal/8c4778817d5b4fad92166a8840486cc8/system@8ae22133f9cc44ce99f6ab56b3cb109c-00000000003ed83f-0005e1aa197ecd14.journal: Journal file, offline
/var/log/journal/8c4778817d5b4fad92166a8840486cc8/system@8ae22133f9cc44ce99f6ab56b3cb109c-00000000003f9fae-0005e1aa51d006dd.journal: Journal file, offline
/var/log/journal/8c4778817d5b4fad92166a8840486cc8/system@8ae22133f9cc44ce99f6ab56b3cb109c-000000000041c7f8-0005e1aaf01272b3.journal: Journal file, archived
/var/log/journal/8c4778817d5b4fad92166a8840486cc8/system.journal:                                                                    Journal file, online
```

There's a few "offline" journals and only one "archived" ones. Each one is 8M which is smaller than 10M but combined they are 48M. This setting doesn't seem to work because user-specific journals are 40M+ **each**. And `journalctl` **knows this**:
```
root@norns:~# journalctl --disk-usage
Archived and active journals take up 315.2M in the file system.
```

## Workaround for now
This removes the old journal files.
```
$ journalctl --verify
$ journalctl --vacuum-size=10M
```

tehn, 2022-06-18:
have you identified any misconfiguration?

catfact, 2023-11-21:
sorry for all the clicks, i'll open a new one

closed 2023-11-21 by catfact