monome/norns-image #103: daemon.log shouldn't be filling up

catfact, 2023-11-21:
this is like issue #99 , but the problem isn't printing on OSC rx or any other particular decision to log something. daemon.log shouldn't be growing to huge proportions and probably shouldn't be growing at all.

here's a forum post i wrote up:
https://llllllll.co/t/storage-issues-on-norns/62816/14?u=graymazes

TL/DR:

- this link was helpful: https://forums.raspberrypi.com/viewtopic.php?t=341605
- AFAICT, systemd doesn't write anything outside of `/var/log/journal`, this thing is written by syslog
- i think it is addressed for me by adding this to `/etc/systemd/journald.conf`:
```
ForwardToConsole=no
ForwardToWall=no
```
- i think this is new behavior in Bullseye, is why it got us

catfact, 2023-11-21:
now that i think about it, for personal use on stock norns, i actually want to turn off logging to disk entirely.

tehn, 2023-11-21:
i think that's pretty legit!

simonvanderveldt, 2023-11-22:
FYI you can probably just uninstall rsyslog https://www.debian.org/releases/bookworm/amd64/release-notes/ch-information.en.html#changes-to-system-logging but do check if it doesn't break anything ;)
I don't know why it was added/enabled/why it showed up now when it didn't before.

catfact, 2023-11-23:
just as another datapoint, after running the rude printer script for a while, i got another huge file at `/var/log/user.log`. (guess i launched the norns process instead of having systemd do it, sorry i can't recall.)

stuff under my `/var/log/journal/` had also grown very large (approaching 2GB.) this despite having a size limit set in `/etc/systemd/journald.conf`.

as an experiment, i did the following for myself:
- completely uninstalled `rsyslog` (thanks simon)
- changed `Storage=persistent` to `Storage=volatile` in `/etc/systemd/journald.conf`
- nuked `/var/log/journal`
- truncated `/var/log/daemon.log` 

results are:
- everything in nornsland seems to be working normally 
- i can still see service output with `journalctl`
- `daemon.log` and `user.log` are still zeroed
- `/var/log/journal` still doesn't exist
- stome stuff is still being created like `auth.log` and `kern.log`

...so far i'd feel better about running this way on stock norns.

i guess the thing that bothers me the most is how / why these size limits don't seem to be applied. (there is a size limit set in `/etc/rsyslog.d/rsyslog`, but that didn't seem to matter.)

hvij1228, 2023-11-24:
Is  OS ver. 231114 compatible with pre-211028 shields (raspbi 3)? I had to do a clean install  to OS ver. 220306 and when I updated to 231114 my norns shield when restarted was blank but was generating the boot-up soundbyte. My norns unit works as expected otherwise. I'm just unable to run some of the scripts when installed (error: load fail).

tehn, 2024-02-14:
i've modified the update script to remove rsyslog and the log files. i've also rewritten the journald config.

tested locally, it works.

this change will go into the next update.

thank you!

closed 2024-02-14 by tehn